nwesem/mtcnn_facenet_cpp_tensorRT #1: Cant make MTCNN engines

yomuevans, 2019-11-22:
Hi, so i followed step by step instruction, but the build cant seem to make mtcnn model engines by itself, here's the output am getting

Start generating mtCNN TenosrRT runtime models
changedH = 96, changedW = 128
rawName = ../mtCNNModels/det1_relu1.engine
[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format ditcaffe.NetParameter: 7:1: Expected identifier, got: <
ERROR: CaffeParser: Could not parse deploy file
Segmentation fault (core dumped)

"det1_relu1" i had succesfully created my own engines placed them in the folder, apparently the code seems to want to make its own hence adding '1' at the end the name

nwesem, 2019-11-22:
Hi yomuevans, it could work when you created your custom engines, but the [base_engine.cpp](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/master/src/baseEngine.cpp) takes care of that. So, as stated in the [README.md](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/master/README.md), you will only need the models from [PKUZHOU's repo](https://github.com/PKUZHOU/MTCNN_FaceDetection_TensorRT/). 

In this case freezing and converting the model to uff only holds for the Tensorflow model of the FaceNet (facenet.pb --> facenet.uff).
Can you try again without creating custom .engine files, but just put the facenet.uff in the [facenetModels](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/tree/master/facenetModels) and all the mtCNN models in the [mtCNNModels](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/tree/master/mtCNNModels) folder?

yomuevans, 2019-11-27:
Hi again, facenet conversion to .engine works fine. Am still getting the same error, even after formatting the whole disk and reinstalling the dependancies.

Can i somehow disable the base.cpp? Coz all i need now is just run inference. Converting all the the models isnt the issue for me. I just need the code to execute. Kindly help.

nwesem, 2019-11-27:
1. You do not have to convert the models for the mtCNN. The mtCNN folder structure should look like this:
det1_relu.caffemodel
det1_relu.prototxt
det2_relu.caffemodel
det2_relu.prototxt
det3_relu.caffemodel
det3_relu.prototxt
README.md

If you have all the files in the **mtCNN folder** you should be able to run the entire project using `./mtcnn_facenet_cpp_tensorRT`. When run for the first time the base_engine.cpp will create several _*.engine_ files.
 
2. You will need TensorRT 5.1 or higher

Hope it works now

yomuevans, 2019-12-03:
Thanks it worked, but i didnt use PKUZHOU's repo for the MTCNN models. but now i got new problem. With no images in the img folder am getting this output.

End generating TensorRT runtime models
Parsing Directory: ../imgs
Empty frame! Exiting...
Counted 0 frames in 0 seconds! This equals nanfps.
FaceNet was destructed

With an image any .jpg image(even after renaming an image to *class_name*.jpg) this is the output

End generating TensorRT runtime models
Parsing Directory: ../imgs
OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /home/nvidia/build_opencv/opencv/modules/core/src/matrix.cpp, line 538
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/nvidia/build_opencv/opencv/modules/core/src/matrix.cpp:538: error: (-215) 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows in function Mat

Aborted (core dumped)

yomuevans, 2019-12-04:
So everything worked, everything, but i had to make lots of changes to the code. If you told me to repeat the changes ive made id probably fail spectacularly, currently i have no idea. But Thanks you've really helpful

closed 2019-12-04 by yomuevans

deaffella, 2019-12-18:
> Hi, so i followed step by step instruction, but the build cant seem to make mtcnn model engines by itself, here's the output am getting
> 
> Start generating mtCNN TenosrRT runtime models
> changedH = 96, changedW = 128
> rawName = ../mtCNNModels/det1_relu1.engine
> [libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format ditcaffe.NetParameter: 7:1: Expected identifier, got: <
> ERROR: CaffeParser: Could not parse deploy file
> Segmentation fault (core dumped)
> 
> "det1_relu1" i had succesfully created my own engines placed them in the folder, apparently the code seems to want to make its own hence adding '1' at the end the name
> Can i run the code with my custom made engines?

hi! I ran into the same problem. 
how did you load the face detector models? I just try to execute step01_pb_to_uff.py, then download models with terminal by links in readme and run ./mtcnn_facenet_cpp_tensorRT
but every time i've got an error like you

yomuevans, 2019-12-18:
Dont use wget to download the models

nwesem, 2019-12-18:
Thanks @yomuevans, I will update the README.md according to that.