nwesem/mtcnn_facenet_cpp_tensorRT #2: Using model using TensorRT and Python

julayelei, 2020-01-28:
Hello! I've just tested this C++ implementation and is really great! Thank you for sharing your knowledge.
I was wondering if this model could be used in Python with TensorRT.
Thanks in advance!

nwesem, 2020-01-28:
Hi julaylei, first of all thanks a lot I appreciate it!
Of course the models can likewise be used in Python using TensorRT, but it would probably take some time to reimplement the preprocessing using Python. I guess that would be quite good practice though ;-). Also, it should be pretty straight forward since the Python API of OpenCV is very close to the C++ API.
Does it make a huge difference for you? Why do you need it to run in Python? 
Cheers

nwesem, 2020-02-26:
#6

closed 2020-02-26 by nwesem