nwesem/mtcnn_facenet_cpp_tensorRT #3: No opencv window comes up after all the steps

zerocool95, 2020-02-06:
I am trying to run the code on jetson nano and everything works like a charm but in the end i dont see any screen , only some continuous logs on the screen . What could be the possible reason for this?
I have also changed the camera to usb camera and tested it on jetson via a python code and the camera seems to work perfectly fine. Any help would be appreciated !

Thanks !

nwesem, 2020-02-07:
hi zerocool95, usually in OpenCV the combination of `cv::imshow("VideoSource", frame);` and `char keyboard = cv::waitKey(1);` should create and update the OpenCV GUI. As this is what happens in the `main.cpp`, it should show the GUI or as you call it the window. You can find it in the `main.cpp` here: https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/master/src/main.cpp#L88-L93

i am guessing that you have changed one of these lines in the code? let me know if that helps, if not can you fork the repository and push your current code?

zerocool95, 2020-02-07:
I haven't changed any of the code in the current repository. Is it possible for it to a memory issue because i dont have much memory available on my jetson?

lucasBruge, 2020-02-13:
I think I may have the same kind of problem... It runs, but only writes INFOs to the console, like

`INFO: tatic 0 time0.213432648 `

 ...I added some debug outputs to main.cpp, and it seems trapped on FaceNetClassifier() initialization on line 41.

 But I just realized that I don't have any "facenet.engine" file (line 28)... Of course it's my bad, but I'm lost...

 I'll appreciate some help... thanks!!! Great work!!!

lucasBruge, 2020-02-13:
...I woked!!! It was creating the files!!! It needed some patience... great!!!

lucasBruge, 2020-02-13:
...but It didn't a second time :(

do-van-long, 2020-03-30:
@lucasBruge me too. Have you found a solution yet?
@nwesem Could you please help me with the second execution? The error looks like this: 
=========================================
Empty frame! Exiting ...
Counted 0 frames in 0 seconds! This equals nanfps.
Segmentation fault (core dumped)
=========================================

lucasBruge, 2020-03-30:
Ey, @do-van-long. It was kind of strange thing I couldn't figure it out. It started to work ok just wright after the boot. But, it keeps failing sometimes then... I mean, sometimes works, the second, ther or forth time... but sometimes it didn't.... don't know why

do-van-long, 2020-03-31:
@lucasBruge Thank for your answer. I think it's the camera problem. After resetting my Jetson Nano and check for camera, the program works quite well.

sivak95, 2020-04-01:
Has anyone got this to work using OpenCV 4.2?

lucasBruge, 2020-04-02:
Check this comment (and the thread), it may help...

https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/issues/4#issuecomment-585973312

nwesem, 2020-04-02:
Thanks again @lucasBruge. It seems like multiple people have had this issue. Does it make sense to change this on the master branch? What do you think @lucasBruge ?

git-yu-inoue, 2020-05-21:
@lucasBruge, do-van-long
second time problem's Cause is not release VideoStreamer.

src/videoStreamer.cpp
add file end
---
void VideoStreamer::release() {
    m_capture->release();
}
---
src/videoStreamer.h
add this
---
...
public
       std::string gstreamer_pipeline (int capture_width, int capture_height, int display_width, int   display_height, int frameRate, int flip_method=0);
    void release(); // <- this line add
};
---

src/main.cpp
add this
---
    auto globalTimeEnd = chrono::steady_clock::now();
    videoStreamer.release(); // <- this line add
    cv::destroyAllWindows();
---

I worked.

and opencv problem is there
CV_RGB2BGR -> cv::COLOR_RGB2BGR
---
       //cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, CV_RGB2BGR);
        cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, cv::COLOR_RGB2BGR);
---

nwesem, 2020-09-18:
Thanks a lot! All of this was fixed and released on master branch, also there are tags now so you can jump back if you are using an older version of Jetpack

closed 2020-09-18 by nwesem