nwesem/mtcnn_facenet_cpp_tensorRT #4: Failed to Compile facenet

jamesee, 2020-02-11:
Hi

I followed your github instruction closely, but I had the following errors:

dlinano@jetson-nano:~/mtcnn_facenet_cpp_tensorRT/build$ make -j${nproc}
[  7%] Building NVCC (Device) object trt_l2norm_helper/CMakeFiles/trt_l2norm_helper.dir/trt_l2norm_helper_generated_l2norm_helper.cu.o
Scanning dependencies of target trt_l2norm_helper
[ 14%] Building CXX object trt_l2norm_helper/CMakeFiles/trt_l2norm_helper.dir/l2norm_helper.cpp.o
[ 21%] Linking CXX static library libtrt_l2norm_helper.a
[ 21%] Built target trt_l2norm_helper
Scanning dependencies of target mtcnn_facenet_cpp_tensorRT
[ 28%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/baseEngine.cpp.o
[ 35%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o
[ 42%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/common.cpp.o
[ 50%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/network.cpp.o
[ 57%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/pnet_rt.cpp.o
[ 64%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/main.cpp.o
[ 78%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/mtcnn.cpp.o
[ 78%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/videoStreamer.cpp.o
[ 85%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/onet_rt.cpp.o
[ 92%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/rnet_rt.cpp.o
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp: In member function ‘void FaceNetClassifier::createOrLoadEngine()’:
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:54:62: warning: ‘virtual nvinfer1::INetworkDefinition* nvinfer1::IBuilder::createNetwork()’ is deprecated [-Wdeprecated-declarations]
         INetworkDefinition *network = builder->createNetwork();
                                                              ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:5431:58: note: declared here
     TRT_DEPRECATED virtual nvinfer1::INetworkDefinition* createNetwork() TRTNOEXCEPT = 0;
                                                          ^~~~~~~~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:56:59: warning: ‘DimsCHW’ is deprecated [-Wdeprecated-declarations]
         parser->registerInput("input", DimsCHW(160, 160, 3), UffInputOrder::kNHWC);
                                                           ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:206:22: note: declared here
 class TRT_DEPRECATED DimsCHW : public Dims3
                      ^~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:71:38: warning: ‘virtual void nvinfer1::IBuilder::setFp16Mode(bool)’ is deprecated [-Wdeprecated-declarations]
             builder->setFp16Mode(true);
                                      ^
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp: In member function ‘virtual void baseEngine::caffeToGIEModel(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, nvinfer1::IHostMemory*&)’:
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:71:62: warning: ‘virtual nvinfer1::INetworkDefinition* nvinfer1::IBuilder::createNetwork()’ is deprecated [-Wdeprecated-declarations]
         INetworkDefinition *network = builder->createNetwork();
                                                              ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:5757:33: note: declared here
     TRT_DEPRECATED virtual void setFp16Mode(bool mode) TRTNOEXCEPT = 0;
                                 ^~~~~~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:74:38: warning: ‘virtual void nvinfer1::IBuilder::setInt8Mode(bool)’ is deprecated [-Wdeprecated-declarations]
             builder->setInt8Mode(true);
                                      ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:5595:33: note: declared here
     TRT_DEPRECATED virtual void setInt8Mode(bool mode) TRTNOEXCEPT = 0;
                                 ^~~~~~~~~~~
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/common.h:7:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5431:58: note: declared here
     TRT_DEPRECATED virtual nvinfer1::INetworkDefinition* createNetwork() TRTNOEXCEPT = 0;
                                                          ^~~~~~~~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:79:43: warning: ‘virtual void nvinfer1::IBuilder::setMaxWorkspaceSize(std::size_t)’ is deprecated [-Wdeprecated-declarations]
         builder->setMaxWorkspaceSize(1<<30);
                                           ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:5462:33: note: declared here
[... 18 lines cut ...]
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5462:33: note: declared here
     TRT_DEPRECATED virtual void setMaxWorkspaceSize(std::size_t workspaceSize) TRTNOEXCEPT = 0;
                                 ^~~~~~~~~~~~~~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:85:64: warning: ‘virtual nvinfer1::ICudaEngine* nvinfer1::IBuilder::buildCudaEngine(nvinfer1::INetworkDefinition&)’ is deprecated [-Wdeprecated-declarations]
         ICudaEngine *engine = builder->buildCudaEngine(*network);
                                                                ^
In file included from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/common.h:7:0,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /home/dlinano/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5566:51: note: declared here
     TRT_DEPRECATED virtual nvinfer1::ICudaEngine* buildCudaEngine(
                                                   ^~~~~~~~~~~~~~~
/home/dlinano/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:128:76: note: suggested alternative: ‘CV_RGB’
         cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, CV_RGB2BGR);
                                                                            ^~~~~~~~~~
                                                                            CV_RGB
CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/build.make:110: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o' failed
make[2]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all' failed
make[1]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

lucasBruge, 2020-02-13:
Hi. Through some google-search, I found that's probably a compatibility issue with opencv (It wasn't related to this code, of course)

 It said that you should try replace CV_RGB2BGR with COLOR_BGR2RGB. I looked in imgproc.hpp, and I found this

COLOR_BGR2RGB = 4,
COLO_RGB2BGR = COLOR_BGR2RGB

 ...so I tried that, but It didn't work. So I replaced it directly by number 4.

 Try this (at faceNet.cpp, line 128)
`cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, 4); // replace CV_RGB2BGR to 4`

 It worked for me... but I know that it isn't the best thing to do...but it may help.

jamesee, 2020-02-17:
@lucasBruge 

Thank you very much. It works. Managed to compile facenet.

nwesem, 2020-02-19:
thanks @lucasBruge

closed 2020-02-19 by nwesem