nwesem/mtcnn_facenet_cpp_tensorRT #5: Parse UFF ERROR

xuwanqi, 2020-02-17:
when i try to parse pb to uff,it shows:
Warning: keepdims is ignored by the UFF Parser and defaults to True
DEBUG: convert reshape to flatten node

and i move the .uff,it shows:
ERROR: InceptionResnetV1/Conv2d_1a_3x3/convolution: kernel weights has count 864 but 46080 was expected
ERROR: UFFParser: Parser error: InceptionResnetV1/Conv2d_1a_3x3/BatchNorm/batchnorm/mul: The input to the Scale Layer is required to have a minimum of 3 dimensions.
Failed to parse UFF

do you have any idea with it?

nwesem, 2020-04-08:
have you found out what the reason for this error is? @xuwanqi

nwesem, 2020-09-18:
1. Check the path to the input file, it is hardcoded in the python script
2. Try using tensorflow 1.15 and python 3.6, I have never had any issues doing it this way and I just tested this again

closed 2020-09-18 by nwesem