nwesem/mtcnn_facenet_cpp_tensorRT #6: Error loading model  with python

elisabetta496, 2020-02-26:
Hi,
I tested your C++ implementation and I would like to implement it in  Python. I'm trying to load the engine file but the problem is that the plugin is not found when loading the engine with:

`with open(engineFile, "rb") as f, trt.Runtime(G_LOGGER) as runtime:
	engine = runtime.deserialize_cuda_engine(f.read())`

I get the following error:

`[TensorRT] ERROR: INVALID_ARGUMENT: getPluginCreator could not find plugin L2Norm_Helper_TRT version 1
[TensorRT] ERROR: safeDeserializationUtils.cpp (259) - Serialization Error in load: 0 (Cannot deserialize plugin since corresponding IPluginCreator not found in Plugin Registry)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
`

Do you know how could I solve  this issue? 
Thanks

nwesem, 2020-02-26:
hi, i'm happy to hear that.
As stated in the README, this is a custom plugin written by https://github.com/r7vme/tensorrt_l2norm_helper. 
Please check how to load custom TensorRT plugins in Python and let me know if you need more help.

deaffella, 2020-03-24:
> Hi,
> I tested your C++ implementation and I would like to implement it in Python. I'm trying to load the engine file but the problem is that the plugin is not found when loading the engine with: [...]

Hi! 
I want to use uff models in my python script, but I don't know how to load them correctly.
Have you solved the problem of loading models in python?

nwesem, 2020-03-24:
does [this part](https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#add_custom_layer_python) of TensorRT documentation help you?

deaffella, 2020-03-25:
> does [this part](https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#add_custom_layer_python) of TensorRT documentation help you?

No, it does not. I don't see any instructions to load .uff and .engine models in this documentation.

elisabetta496, 2020-03-31:
No, i'm sorry! I've loaded the engine file so far.
Have you solved the problem? Otherwise I will let you know if I will be able.

deaffella, 2020-04-04:
> No, i'm sorry! I've loaded the engine file so far.
> Have you solved the problem? Otherwise I will let you know if I will be able.

Unfortunately I couldn't find any example of loading models in Python. 
Please help me

do-van-long, 2020-05-12:
Hi @elisabetta496!
I have the same problem as @deaffella . Could you please share with us how to convert and save facenet.uff to facenet.engine using Python?
Many thanks.

deaffella, 2020-08-04:
> does [this part](https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#add_custom_layer_python) of TensorRT documentation help you?

hi! I'm still trying to open optimized models in Python. could you help me with advice on how to create a. so file for the plugin?

nwesem, 2020-09-18:
I'm not sure if this is the way it works. Are you @deaffella ? If so, I could help you compile the project as a dynamic or static library.

AnasMK, 2020-10-08:
Check this repo [riotu-lab/tf2trt_with_onnx](https://github.com/riotu-lab/tf2trt_with_onnx) to covert Facenet model to TensorRT engine and use it with Python