nwesem/mtcnn_facenet_cpp_tensorRT #7: Failed to parse UFF

swetha-sree, 2020-03-05:
--------------------------------------------------------------------------------
ERROR: UffParser: Unsupported number of graph 0
Failed to parse UFF
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted (core dumped)
---------------------------------------------------------------------------------

Hi above is the stack trace for the error while executing ./mtcnn_facenet_cpp_tensorRT
Hardware: Jetson Nano with Jetpack 4.2.1
Tensorflow Version: Tried both 1.14 and 1.15 
TensorRT: 5.1.6.1

PhilipCNel, 2020-03-18:
@swetha-sree did you remember to move the FaceNet.uff to the FaceNet models folder?

luanpham107, 2020-03-19:
I tried tensorRT 1.15 and it working great. Please follow Nvidia guidelines for install tensorRT here:
Installation guide - [](https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html 5
)Release note - [](https://docs.nvidia.com/deeplearning/dgx/install-tf-xavier-release-notes/index.html)

mel0615, 2020-04-20:
Hi I had the same error while executing mtcnn_facenet_cpp_tensorRT. Finally I found that I compiled the project under the _root_ directory but not _build_ dir, so that the build files and executable file were in the _root_ dir. Please check the path. Hope it helps. 
TF: 1.14
TRT: 5.1.5.0

nwesem, 2020-09-18:
Try new release [jp4.4](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/releases/tag/jp4.4) with Jetpack 4.4 and tensorflow 1.15