nwesem/mtcnn_facenet_cpp_tensorRT #9: Cuda 10.2

celevra, 2020-04-23:
Hi,

got my Jetson nano yesterday and i get following errors:
```
2020-04-23 20:54:33.340662: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcudart.so.10.0'; dlerror: libcudart.so.10.0: cannot open shared object file: No such file or directory
2020-04-23 20:54:33.340727: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
2020-04-23 20:54:36.024791: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer.so.6'; dlerror: libnvinfer.so.6: cannot open shared object file: No such file or directory
2020-04-23 20:54:36.024987: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer_plugin.so.6'; dlerror: libnvinfer_plugin.so.6: cannot open shared object file: No such file or directory
2020-04-23 20:54:36.025035: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:30] Cannot dlopen some TensorRT libraries. If you would like to use Nvidia GPU with TensorRT, please make sure the missing libraries mentioned above are installed properly.
NOTE: UFF has been tested with TensorFlow 1.14.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.7
=== Automatically deduced input nodes ===
[name: "input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
    }
  }
}
]
=========================================

Using output node embeddings
Converting to UFF graph
Warning: No conversion function registered for layer: L2Norm_Helper_TRT yet.
Converting embeddings/Rsqrt as custom op: L2Norm_Helper_TRT
Traceback (most recent call last):
  File "step01_pb_to_uff.py", line 33, in <module>
    output_filename=uff_file, text=False)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py", line 178, in from_tensorflow
    debug_mode=debug_mode)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 94, in convert_tf2uff_graph
    uff_graph, input_replacements, debug_mode=debug_mode)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 79, in convert_tf2uff_node
    op, name, tf_node, inputs, uff_graph, tf_nodes=tf_nodes, debug_mode=debug_mode)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 41, in convert_layer
    fields = cls.parse_tf_attrs(tf_node.attr)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 222, in parse_tf_attrs
    return {key: cls.parse_tf_attr_value(val) for key, val in attrs.items() if val is not None and val.WhichOneof('value') is not None}
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 222, in <dictcomp>
    return {key: cls.parse_tf_attr_value(val) for key, val in attrs.items() if val is not None and val.WhichOneof('value') is not None}
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 218, in parse_tf_attr_value
    return cls.convert_tf2uff_field(code, val)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 179, in convert_tf2uff_field
    if isinstance(val, tf.AttrValue):
AttributeError: module 'tensorflow' has no attribute 'AttrValue'
```

regards

nwesem, 2020-04-28:
@celevra does the current JetPack version for Jetson Nano ship with Cuda 10.2 now? That would mean i have to test everything from scratch

celevra, 2020-04-28:
Yes, nothing Changed from my Side, thats the Default now

Sent from Nine<http://www.9folders.com/>

nwesem, 2020-04-28:
I just checked JetPack 4.4 ships with Cuda 10.2, but it is still in Developer Preview. As soon as they release a production release I will test it with. Until then I would suggest that you download JetPack 4.3.

celevra, 2020-04-28:
Hi,

Oh i've followed the get Starter Guide from Nvidia. Can your Point me in the right direction?

Regards

Sent from Nine<http://www.9folders.com/>

nwesem, 2020-04-28:
Yes follow [this link](https://developer.nvidia.com/jetpack-43-archive) to download sd card image for jetson nano and flash it to your sd card using etcher or startup disk creator on ubuntu

celevra, 2020-04-28:
thank you, i've now installed 4.3 but got an error:

```
ki mtcnn_facenet_cpp_tensorRT # jetson_release
 - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.3 [L4T 32.3.1]
   * NV Power Mode: MAXN - Type: 0
   * jetson_clocks service: inactive
 - Libraries:
   * CUDA: 10.0.326
   * cuDNN: 7.6.3.28
   * TensorRT: 6.0.1.10
   * Visionworks: 1.6.0.500n
   * OpenCV: 4.1.1 compiled CUDA: YES

ki mtcnn_facenet_cpp_tensorRT # python3 step01_pb_to_uff.py
2020-04-28 23:13:55.841645: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
2020-04-28 23:13:58.633144: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libnvinfer.so.6
2020-04-28 23:13:58.634831: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libnvinfer_plugin.so.6
Traceback (most recent call last):
  File "step01_pb_to_uff.py", line 2, in <module>
    import graphsurgeon as gs
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/__init__.py", line 9, in <module>
    from graphsurgeon.StaticGraph import *
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py", line 7, in <module>
    from graphsurgeon._utils import _regex_list_contains_string, _generate_iterable_for_search, _clean_input_name
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/_utils.py", line 2, in <module>
    from tensorflow import NodeDef
ImportError: cannot import name 'NodeDef'
```

i've installed tensorflow from this docu:
```
https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html
```

celevra, 2020-04-28:
ok its Tensorflow 2.1 i should downgrade...

nwesem, 2020-04-28:
yes, tensorflow 1 should work

celevra, 2020-04-29:
now i get this build error:

```
ki build # make -j${nproc}
[  7%] Building NVCC (Device) object trt_l2norm_helper/CMakeFiles/trt_l2norm_helper.dir/trt_l2norm_helper_generated_l2norm_helper.cu.o
Scanning dependencies of target trt_l2norm_helper
[ 14%] Building CXX object trt_l2norm_helper/CMakeFiles/trt_l2norm_helper.dir/l2norm_helper.cpp.o
[ 21%] Linking CXX static library libtrt_l2norm_helper.a
[ 21%] Built target trt_l2norm_helper
Scanning dependencies of target mtcnn_facenet_cpp_tensorRT
[ 28%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/videoStreamer.cpp.o
[ 35%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/main.cpp.o
[ 42%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/baseEngine.cpp.o
[ 50%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/rnet_rt.cpp.o
[ 57%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/network.cpp.o
[ 64%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/common.cpp.o
[ 71%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o
[ 78%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/pnet_rt.cpp.o
[ 85%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/onet_rt.cpp.o
[ 92%] Building CXX object CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/mtcnn.cpp.o
/root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp: In member function ‘virtual void baseEngine::caffeToGIEModel(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, nvinfer1::IHostMemory*&)’:
/root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:71:62: warning: ‘virtual nvinfer1::INetworkDefinition* nvinfer1::IBuilder::createNetwork()’ is deprecated [-Wdeprecated-declarations]
         INetworkDefinition *network = builder->createNetwork();
                                                              ^
In file included from /root/mtcnn_facenet_cpp_tensorRT/src/common.h:7:0,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5431:58: note: declared here
     TRT_DEPRECATED virtual nvinfer1::INetworkDefinition* createNetwork() TRTNOEXCEPT = 0;
                                                          ^~~~~~~~~~~~~
/root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:84:45: warning: ‘virtual void nvinfer1::IBuilder::setMaxWorkspaceSize(std::size_t)’ is deprecated [-Wdeprecated-declarations]
         builder->setMaxWorkspaceSize(1 << 25);
                                             ^
In file included from /root/mtcnn_facenet_cpp_tensorRT/src/common.h:7:0,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5462:33: note: declared here
     TRT_DEPRECATED virtual void setMaxWorkspaceSize(std::size_t workspaceSize) TRTNOEXCEPT = 0;
                                 ^~~~~~~~~~~~~~~~~~~
/root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:85:64: warning: ‘virtual nvinfer1::ICudaEngine* nvinfer1::IBuilder::buildCudaEngine(nvinfer1::INetworkDefinition&)’ is deprecated  -Wdeprecated-declarations]
         ICudaEngine *engine = builder->buildCudaEngine(*network);
                                                                ^
In file included from /root/mtcnn_facenet_cpp_tensorRT/src/common.h:7:0,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.h:4,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/baseEngine.cpp:5:
/usr/include/aarch64-linux-gnu/NvInfer.h:5566:51: note: declared here
     TRT_DEPRECATED virtual nvinfer1::ICudaEngine* buildCudaEngine(
                                                   ^~~~~~~~~~~~~~~
/root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp: In member function ‘void FaceNetClassifier::createOrLoadEngine()’:
/root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:54:62: warning: ‘virtual nvinfer1::INetworkDefinition* nvinfer1::IBuilder::createNetwork()’ is deprecated [-Wdeprecated-declarations]
         INetworkDefinition *network = builder->createNetwork();
                                                              ^
In file included from /root/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:5431:58: note: declared here
     TRT_DEPRECATED virtual nvinfer1::INetworkDefinition* createNetwork() TRTNOEXCEPT = 0;
                                                          ^~~~~~~~~~~~~
/root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:56:59: warning: ‘DimsCHW’ is deprecated [-Wdeprecated-declarations]
         parser->registerInput("input", DimsCHW(160, 160, 3), UffInputOrder::kNHWC);
                                                           ^
In file included from /root/mtcnn_facenet_cpp_tensorRT/src/faceNet.h:15:0,
                 from /root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:1:
/usr/include/aarch64-linux-gnu/NvInfer.h:206:22: note: declared here
 class TRT_DEPRECATED DimsCHW : public Dims3
                      ^~~~~~~
/root/mtcnn_facenet_cpp_tensorRT/src/faceNet.cpp:71:38: warning: ‘virtual void nvinfer1::IBuilder::setFp16Mode(bool)’ is deprecated [-Wdeprecated-declarations]
             builder->setFp16Mode(true);
                                      ^
[... cut ...]
CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/build.make:110: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o' failed
make[2]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all' failed
make[1]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

nwesem, 2020-04-29:
known issue due to OpenCV 4.2 --> https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/issues/4#issuecomment-585973312

I will update this asap.

git-yu-inoue, 2020-05-23:
//cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, CV_RGB2BGR);
    cv::cvtColor(m_croppedFaces[i].faceMat, m_croppedFaces[i].faceMat, cv::COLOR_RGB2BGR);

nwesem, 2020-09-18:
Cuda 10.2 works on develop branch, can anybody check if everything is properly working?

nwesem, 2020-09-18:
Tested and merged to master branch also there are tags now for different Jetpack versions

closed 2020-09-18 by nwesem