nwesem/mtcnn_facenet_cpp_tensorRT #10: Error with facenet.pb to facenet.uff

nagorntow, 2020-04-26:
Using output node dimension/LeakyRelu
Using output node orientation/l2_normalize
Using output node confidence/Softmax
Converting to UFF graph
Traceback (most recent call last):
  File "./step01_pb_to_uff.py", line 32, in <module>
    output_filename=uff_file, text=False)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py", line 178, in from_tensorflow
    debug_mode=debug_mode)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 94, in convert_tf2uff_graph
    uff_graph, input_replacements, debug_mode=debug_mode)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py", line 62, in convert_tf2uff_node
    raise UffException(str(name) + " was not found in the graph. Please use the -l option to list nodes in the graph.")
uff.model.exceptions.UffException: confidence/Softmax was not found in the graph. Please use the -l option to list nodes in the graph.

nwesem, 2020-04-28:
@nagorntow can you give more info what exactly you are using as your input file?

nagorntow, 2020-04-28:
I using file download from github.com/apollo-time/facenet/raw/master/model/resnet/facenet.pb and change name file to sample.pb

nwesem, 2020-04-28:
did you change the input path (variable is called `pb_file`) to `path/to/your/sample.pb` in [this line](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/master/step01_pb_to_uff.py#L10)?

nwesem, 2020-09-18:
I tested this with Jetpack 4.4 and it is properly working with tensorflow 1.15 and python 3.6

closed 2020-09-18 by nwesem