nwesem/mtcnn_facenet_cpp_tensorRT #11: Question of inputsize

mel0615, 2020-05-11:
hi I had an error while running this program and I wonder if there's any solution. 
The _setting-inputsize_ of the model are _640_ and _480_, and the _minsize_ of mtcnn is _60_. But when I change the two parameters, the program doesnot run successfully. 
According to the model, it build a image pyramid so that any size of images should work. Is there any rule when setting the parameters? I check the program and cannot find the reason so I wonder if you could give some advices. Thanks a lot.

closed 2020-05-12 by mel0615

nwesem, 2020-05-12:
Hey @mel0615, what was the problem?

mel0615, 2020-05-13:
@nwesem Oh it's a small mistake. 
Finally I found that the program didnot create a NEW trt-engine when it built with a new group of parameters. It found the path of the PRIOR engine created by prior parameters, and loaded it. I just need to recreate it. 
Thank u anyway. I should be more careful, lol.