nwesem/mtcnn_facenet_cpp_tensorRT #12: Failed to convert facenet.pb model to facenet.onnx model

do-van-long, 2020-05-12:
Hi @nwesem,
Since converting a tensorflow model to UFF model will be deprecated in the future and transforming deep learning models to ONNX is becoming popular, I have tried to convert facenet.pb to facenet.onnx using tf2onnx but I have not succeeded. The problem is that there are some layers in facenet.pb that are not supported in tf2onnx.
Have you ever tried to do that? or Could anyone give me some suggestions on how to DEBUG? One of the BUGs is 
`File "/home/vanlong/.virtualenvs/facerecog/lib/python3.6/site-packages/tensorflow/python/framework/importer.py", line 430, in import_graph_def
    raise ValueError(str(e))
ValueError: Shape must be rank 4 but is rank 0 for 'InceptionResnetV1/Conv2d_1a_3x3/BatchNorm/batchnorm/mul' (op: 'Conv2D') with input shapes: [], [3,3,3,32].
`

nwesem, 2020-06-01:
Have you found a solution to this issue yet @do-van-long ?

AnasMK, 2020-10-01:
I have successfully converted Facenet to ONNX model. But, the facenet.pb file provided by this repo didn't work.
I used [facenet_keras.h5](https://drive.google.com/drive/folders/1pwQ3H4aJ8a6yyJHZkTwtjcL4wYWQb7bn) provided in this [tutorial](https://machinelearningmastery.com/how-to-develop-a-face-recognition-system-using-facenet-in-keras-and-an-svm-classifier/) and convert it to ONNX and then to TRT using [this tutorial](https://developer.nvidia.com/blog/speeding-up-deep-learning-inference-using-tensorflow-onnx-and-tensorrt/) provided by Nvidia

AnasMK, 2020-10-08:
Check this repo [riotu-lab/tf2trt_with_onnx](https://github.com/riotu-lab/tf2trt_with_onnx) to covert Facenet model to TensorRT engine and use it with Python