nwesem/mtcnn_facenet_cpp_tensorRT #13: cmake error makefile all failed

glennford49, 2020-06-01:
CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/build.make:110: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o' failed

make[2]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/src/faceNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all' failed
make[1]: *** [CMakeFiles/mtcnn_facenet_cpp_tensorRT.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

nwesem, 2020-06-01:
On what kind of machine are you trying to build this project?

glennford49, 2020-06-01:
Jetson nano 4.3 jetpack ,any idea on this error?

nwesem, 2020-09-18:
@glennford49 try building the develop branch, it should work for Jetpack 4.4. I will merge this as soon i have fixed all minor bugs

closed 2020-09-18 by nwesem