nwesem/mtcnn_facenet_cpp_tensorRT #16: facenet is not accurate

xudh1991, 2020-07-07:
I have run through the whole process, but the facenet is not accurate. 
I tested the accuracy of face recognition, but the accuracy rate is only 20%.
I suspect there are some errors between the image and the model, such as normalization, color change, I tried to find the error but no success
I looked at the feature vector, the whole value is very small, most are less than 0.1
I need help, please give me some suggestions， thanks

gugumituo, 2020-08-12:
Hi,have you solve the problem?I also meet the same issue.
I found when i put two same pictures to the model,the distance is 0;
But i put a cropped image from the old one,the distance becomes a big value,which is bigger than 1000.
I don't know why,please help me!

lnxguyz, 2020-11-02:
I am facing the same issue, pls help

xudh1991, 2020-11-03:
> Hi,have you solve the problem?I also meet the same issue.
> I found when i put two same pictures to the model,the distance is 0;
> But i put a cropped image from the old one,the distance becomes a big value,which is bigger than 1000.
> I don't know why,please help me!

xudh1991, 2020-11-03:
I've given up the project. I suggest giving up

closed 2020-11-03 by xudh1991