nwesem/mtcnn_facenet_cpp_tensorRT #17: Core at changing video stream from cam to rstp

githubutilities, 2020-09-26:
I just change cam to rstp videostream, and I got a core dump, which I trace it back to https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/master/src/mtcnn.cpp#L85. Any idea how to fix it?

nwesem, 2020-09-27:
hi githubutilities, using RSTP videostreams as an input would be a cool enhancement. Can you fork the repository and push your code such that I can take a look? Maybe I can make it work

githubutilities, 2020-10-02:
It's simple. I just change line `VideoStreamer videoStreamer = VideoStreamer("rtsp://192.168.10.21:8554/unicast", videoFrameWidth, videoFrameHeight);` in file `main.cpp`. This can get frame from rstp video, but core at feeding the frame into facenet model.

closed 2020-10-24 by nwesem

jishang1602, 2020-12-24:
i tried with the VideoStreamer videoStreamer = VideoStreamer("rtsp://192.168.10.21:8554/unicast", videoFrameWidth, videoFrameHeight); in main.cpp but it is getting error which is shown below.

 
End generating TensorRT runtime models
Parsing Directory: ../imgs
Segmentation fault (core dumped)

in videoStreamer.h and videoStreamer.cpp
VideoStreamer(int nmbrDevice, int videoWidth, int videoHeight, int frameRate, bool isCSICam);
i changed with 
VideoStreamer(const char* nmbrDevice, int videoWidth, int videoHeight, int frameRate, bool isCSICam);

but getting the error
Segmentation fault (core dumped)

nwesem, 2021-01-11:
@jishang1602 Have you solved this? Maybe running it with gdb will help you find where it crashes. Currently, I don't have a setup to test this.

jishang1602, 2021-01-12:
@nwesem NO, it is still not solved. I have to convert the video to 640x480 size. May be this model is resized to 640x480. I have checked the https://github.com/PKUZHOU/MTCNN_FaceDetection_TensorRT/blob/dfad60565216a68413f434b500168c456fdd2587/src/demo.cpp file also. they are also doing resize of the frame. I need one more help from you. Can you help me in videoStreamer.cpp with threading?  on rtsp  the video is coming delayed. And after processing the delayed is increaded. My rtsp works on 20FPS. and the max FPS from the above project is 15-16 FPS. I need help on the real time processing. On Python i can do the threading and get the latest frame. In c++ i am unable to do  that part. thank you

shubham-shahh, 2021-02-10:
> @nwesem NO, it is still not solved. I have to convert the video to 640x480 size. May be this model is resized to 640x480. I have checked the [...]

how do you resize it?

jishang1602, 2021-02-10:
Once the frame is read from video or image.

E.g.,
frame = cv::imread (image_path);
cv:: resize (frame,frame,cv:: Size (640,480),0,0,cv::INTER_CUBIC);

This way.

shubham-shahh, 2021-02-10:
> Once the frame is read from video or image.
> 
> E.g.,
> frame = cv::imread (image_path);
> cv:: resize (frame,frame,cv:: Size (640,480),0,0,cv::INTER_CUBIC);
> 
> This way.

I am trying to take the input from rtsp stream