nwesem/mtcnn_facenet_cpp_tensorRT #18: changing the screen resolution

han-bun, 2020-10-04:
I am not familiar with opencv so pardon me with my question.

I would like to ask if it is possible to run the program with 720p resolution?
I tried doing that and face detection did not work.
I am currently understanding the code in the facenet.cpp
Can you help me with this?

Thank you.

nwesem, 2020-10-05:
Yes, that's possible, but you will have to rebuild all mtcnn engines (you can simply delete them from your mtcnnModels folder and they will be automatically rebuilt).
You cannot use the same ones because mtcnn builds smaller/higher amount of engines in case you want to scale up/down (respectively). hope that will solve your issue.

han-bun, 2020-10-08:
Thank you for answering my question.
I will try your recommendation.
If I have not succeeded, I will ask a follow up question.
(Sorry I am still learning about deep learning stuffs.)

han-bun, 2020-10-23:
Thank you for your help, I was able to output the program in 720p.

I would like to ask one more question.

I am using web camera and I want to use mjpeg in the video writter so I add the following code.

m_capture->set(cv::CAP_PROP_FOURCC,cv::VideoWriter::fourcc('M','J','P','G') );

but was not successful. The system still uses YUYV pixel format.

Is there any problem that was encountered that has the same as mine and is there a solution to this? 
Thank you.