nwesem/mtcnn_facenet_cpp_tensorRT #19: Unable to find jpg in imgs folder

prakharter, 2020-10-14:
I followed the steps and added person1.jpg and person2.jpg in imgs folder but when i run ./mtcnn_facenet_cpp_tensorRT inside build folder i get the below error , please advise what am i doing wrong , thank you

End generate rnet runtime models
rawName = ../mtCNNModels/det3_relu.engine
size1924528
size1924528UNKNOWN: Deserialize required 44143 microseconds.

End generating TensorRT runtime models
Parsing Directory: ../imgs
Empty frame! Exiting...

nwesem, 2020-10-14:
Hey prakharter, 
the jpgs were successfully read, the actual problem is your video device you are using as input. This repository depends on a video stream input (e.g. a camera device). [This if-clause](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/blob/4ebb525e6e89cfbd6e80cd47d12a4fdb7223f8f6/src/main.cpp#L72-L77) fails in your case. That means the video device you are using provides an invalid / empty frame so there is nothing to read the "known" faces (your jpg files) from. Which device are you using?

prakharter, 2020-10-14:
Thanks so much , yes the problem was with the camera not connected properly though after connection the program is stuck at the below point - 

```
video/x-raw, width=(int)640, height=(int)480, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 4 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
```