nwesem/mtcnn_facenet_cpp_tensorRT #20: Empty frame! Exiting...  Try restarting nvargus-daemon by doing: sudo systemctl restart nvargus-daemon

Darshcg, 2020-11-11:
Hi @nwesem,

Thanks for this Repo. 
I am testing this in my Laptop having RTX 2060. 
When I am running this command `./mtcnn_facenet_cpp_tensorRT` I am getting like this:
**End generating TensorRT runtime models
Parsing Directory: ../imgs
Empty frame! Exiting...
 Try restarting nvargus-daemon by doing: sudo systemctl restart nvargus-daemon
Counted 0 frames in 0 seconds! This equals -nanfps.**

May I know what i am doing wrong?

Thanks in advance

nwesem, 2020-11-11:
https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/issues/19

Darshcg, 2020-11-12:
Hi @nwesem,
Thanks for your reply.
I am running the example in my Laptop(/dev/video0), I also set **isCSICam = false**.
But still I am facing an Issue.

Darshcg, 2020-11-13:
This has been Resolved thank you.

closed 2020-11-13 by Darshcg