nwesem/mtcnn_facenet_cpp_tensorRT #25: Face Registration with multiple faces of a Single Person

Darshcg, 2020-12-16:
Hi @nwesem,

Thank you for your Repo.
I was having a doubt on giving a multiple faces of a Single Person for Registration. How to do so?
If we are giving a single image for the Person the accuracy is not good. So, I thought of giving a multiple images for a Single Person to get the good accuracy.

Thanks in advance
Darshan

shubham-shahh, 2021-02-09:
you need to create a .pkl file with all the facial embedding of a faces(be it single or multiple) of the same person and then compare that embedding with the feature vector generated by facenet. soon ill put up a PR with this feature

Tetsujinfr, 2021-05-03:
hi, any update on adding this feature to the repo?

btw I think the accuracy is quite good when testing with a single image per ID in the DB. It would be amazing to be able to efficiently add more faces for a given ID (instead of adding pictures of the same person in the DB which I guess is not good performance wise)

shubham-shahh, 2021-05-05:
> hi, any update on adding this feature to the repo?
>
> btw I think the accuracy is quite good when testing with a single image per ID in the DB. It would be amazing to be able to efficiently add more [...]

Hi please refer the deepstream issue for updates [here](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/issues/30)