nwesem/mtcnn_facenet_cpp_tensorRT #27: rtsp link is not working

jishang1602, 2020-12-24:
i tried with the VideoStreamer videoStreamer = VideoStreamer("rtsp://192.168.10.21:8554/unicast", videoFrameWidth, videoFrameHeight); in main.cpp but it is getting error which is shown below.

End generating TensorRT runtime models
Parsing Directory: ../imgs
Segmentation fault (core dumped)

in videoStreamer.h and videoStreamer.cpp
VideoStreamer(int nmbrDevice, int videoWidth, int videoHeight, int frameRate, bool isCSICam);
i changed with
VideoStreamer(const char* nmbrDevice, int videoWidth, int videoHeight, int frameRate, bool isCSICam);

but getting the error
Segmentation fault (core dumped)

jishang1602, 2020-12-29:
I have checked with different frame size on camera or CSI camera. In Jetpack 4.4 it works with only 640*480 resolution. If i am using rtsp original resolution (640*360), The code is dump. I changed the size in main.cpp. still the code is dump. Do you know how to work with rtsp any resolution or by default resolution of the rtsp. because if the camera is resized than face identification not work. Could you please help me on this?

IllyaPysarchuk, 2021-09-23:
Did you solve it?