nwesem/mtcnn_facenet_cpp_tensorRT #29: Video file test

hirwa145, 2021-01-13:
Is there a way i can use video file as input instead of image/photo.

jishang1602, 2021-01-21:
yes

hirwa145, 2021-01-21:
how can i deploy it using deepstream?

jishang1602, 2021-01-22:
how can i deploy it using deepstream?
--> Not yet build.

Is there a way i can use video file as input instead of image/photo.?
--> in main.cpp changing  the 45th line from 0 to video.mp4
 
44. // init opencv stuff
45.   VideoStreamer videoStreamer = VideoStreamer("video.mp4", videoFrameWidth, videoFrameHeight);
46.   cv::Mat frame;

I hope i answered your query.
Thanks

hirwa145, 2021-01-22:
Thank you. I really appreciate your help and time.

hirwa145, 2021-02-04:
I tried the solution you provided for using mp4 vido file as input. But i am getting errors

jishang1602, 2021-02-04:
Can you please share the error

hirwa145, 2021-02-04:
`End generating TensorRT runtime models
Parsing Directory: ../imgs
Empty frame! Exiting...
 Try restarting nvargus-daemon by doing: sudo systemctl restart nvargus-daemon
Counted 0 frames in 0 seconds! This equals nanfps.
`

hirwa145, 2021-02-04:
After performing this command `/mtcnn_facenet_cpp_tensorRT
`

jishang1602, 2021-02-04:
Did you put your database in the imgs folder? Also need to resize the video file to 640x480 size. Please do this changes and tell me the results

hirwa145, 2021-02-04:
"Did you put your database in the imgs folder", you mean the image of known person to recognise in the video?

jishang1602, 2021-02-04:
Yes the image database that you want to recognise in the video

jishang1602, 2021-02-04:
This folder contains images of people you would like to recognize. The format of picture should class_name.jpg.

hirwa145, 2021-02-04:
in which folder should i put the video test?

hirwa145, 2021-02-04:
is it possible to use 1080x1920?

hirwa145, 2021-02-04:
Thank you, i managed to load the video successfully, even tho it is showing new person on the face of a person whose face is included in the imgs file

hirwa145, 2021-02-04:
Whilest doing the test, i  realised that smaller faces are not detected, is there a way to solve this?

jishang1602, 2021-02-04:
40 is the minimum pixel size for face detection. You can reduce it. But normally people are taking the smaller size upto 40 pixels

hirwa145, 2021-02-04:
how can i change the pixel size? in which source code file?