nwesem/mtcnn_facenet_cpp_tensorRT #32: Cannot read from USB camera

shubham-shahh, 2021-02-09:
Hi @nwesem 
I am using Jetson nano and followed all the steps you've mentioned in readme
when I run ./mtcnn_facenet_cpp_tensorRT
i am getting the following output i have attached in the .txt file
[iss.txt](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/files/5949526/iss.txt)

shubham-shahh, 2021-02-09:
> Hi @nwesem
> I am using Jetson nano and followed all the steps you've mentioned in readme
> when I run ./mtcnn_facenet_cpp_tensorRT
> i am getting the following output i have attached in the .txt file
> [iss.txt](https://github.com/nwesem/mtcnn_facenet_cpp_tensorRT/files/5949526/iss.txt)

solved it by changing the value in main.cpp line 38 set the booean to false

Can i use multiple usb or rtsp sources?

Tetsujinfr, 2021-05-10:
you need to change the code to make the program more flexible regarding type of inputs. You can connect mpi or usb cameras no problem. actually you can tweak the code to pass a gstreamer string as an argument to the prgram, very flexible this way regarding inputs.

I think you can close this issue btw given you auto-resolved it.