nwesem/mtcnn_facenet_cpp_tensorRT #33: how to train the classifier based on subfolders of different people?

nupnup-goddamn, 2021-02-13:
Hi, i would like to use subfolders in the imgs folder of different people so that i can have the classifier to train on mulitple images of the same person (person A) to better distinguish Person A from a display attack of person A (e.g. picture of person A on a phone display).

How should i modify the main.cpp code under getting embeddings of known faces as well as saving new faces of the same person into particular subfolder of that person in the imgs folder?

shubham-shahh, 2021-02-23:
you need to create a separate program which iterated through the folders of images and save embedding of the at person corresponding to the name and save it in a pickle file.
you need to load this pickle file in your program and compare the embedding in pickle file with the ones received from the model at any instance.

nupnup-goddamn, 2021-03-11:
> you need to create a separate program which iterated through the folders of images and save embedding of the at person corresponding to the name and save it in a pickle file.
> you need to load this [...]

Thanks for your insights! I now know how to iterate the folders using os library and saving the model training as a pkl file. cheers!

closed 2021-03-11 by nupnup-goddamn