nwesem/mtcnn_facenet_cpp_tensorRT #43: ValueError: Expected n_neighbors <= n_samples,  but n_samples = 93, n_neighbors = 100

imSrbh, 2021-07-19:
@shubham-shahh can you look into this?
```
Traceback (most recent call last):
  File "test.py", line 324, in sgie_sink_pad_buffer_probe
    result = predict_using_classifier(faces_embeddings, labels, face_to_predict_embedding)
  File "/opt/nvidia/deepstream/deepstream-5.1/sources/deepstream_python_apps/apps/facenet-python/facenet_utils.py", line 51, in predict_using_classifier
    yhat_class = classifier.predict(samples)
  File "/home/ubuntu/.local/lib/python3.6/site-packages/sklearn/neighbors/_classification.py", line 197, in predict
    neigh_dist, neigh_ind = self.kneighbors(X)
  File "/home/ubuntu/.local/lib/python3.6/site-packages/sklearn/neighbors/_base.py", line 683, in kneighbors
    (n_samples_fit, n_neighbors)
ValueError: Expected n_neighbors <= n_samples,  but n_samples = 93, n_neighbors = 100
[INFO] Classifier Training Done...
```

shubham-shahh, 2021-07-19:
Please provide more context, are you training? Inferencing? C++ versi or or python?

imSrbh, 2021-07-19:
- Inferencing via facent-python.
- generated the embeddings.npz
-  In deepstream_facenet.py I added the sgie_probe code from [this](https://github.com/riotu-lab/deepstream-facenet) to classify the faces from known_embeddings. There is facenet_utils.py. Which is basically a classifier (KNN, SVM).

then I was running that.

shubham-shahh, 2021-07-20:
> * Inferencing via facent-python.
> * generated the embeddings.npz
> * In deepstream_facenet.py I added the sgie_probe code from [this](https://github.com/riotu-lab/deepstream-facenet) to classify the faces from known_embeddings. There is facenet_utils.py. Which is basically a classifier (KNN, SVM).
> 
> then I was running that.
> 
> @shubham-shahh

Hi, It has something to do with your KNN classification, I cannot be sure until I see your code.

thandongtb, 2021-07-21:
@imSrbh I guess the reason comes from your dataset. The number sample in your dataset is too small (only 93 images) but the number of nearest neighbors to query is K = 100. I think you should add more images to the dataset (greater than 100) or reduce the K value (maybe 5, 10, or 20 in this case)

imSrbh, 2021-07-21:
@thandongtb Yes, you are right. I was able to resolve it.

closed 2021-07-21 by imSrbh